kirberich/gerber_to_scad #4: How to specify source file

REHolt, 2017-01-28:
I would ike to make use of your software to be able to print solder stencil's for some of my projects. I'm kind of new to using PYTHON on the PC and can't tell from you files, how to specify the source files. 

I was able to generate the output files from my Eagle CAD layout using the .cam file you provided; however, once I have the files, I'm not sure how to tell the Python program to use them. 

I tried using the online version, but get redirected to an APACHE2 error page.

Thanks for creating this, if I can get it to work, it will be a big help with building my boards.

Robert

kirberich, 2017-01-29:
Hi! Yeah sorry about the online version, I've been meaning to get it back up for ages. I'm also sorry for the lack of documentation, I've added a readme now with basic installation and usage instructions: https://github.com/kirberich/gerber_to_scad/blob/master/README.md

Let me know if anything in there is wrong or confusing and I'll try to help! Let me know how you get on, I'd love to see what you make with it!

REHolt, 2017-01-29:
Hey, thanks for the response. I will give this a try today. 

I have been building some ESP8266 projects. I had a PC board made for one of them, but have not been very successful in building it. Seems my solder skills have greatly deteriorated and these SM parts are a pain to hand solder. I decided to try reflow solder and thought it would be great to make my own solder stencil.

I will post back here once I give it a try and let you know how it went.

REHolt, 2017-01-29:
Forgive me, but I guess it's not quite basic enough for me. At the start of your instructions, are you running in a windows command prompt, or are those Linux commands? if it is Linux, can I run this on my Raspberry Pi?

REHolt, 2017-01-29:
I went ahead and tried to set this up on my Raspberry Pi using Raspbian, as that looked more appropriate. It didn't understand the  "virtualenv env", or  the "source env/bin/activate"  (more likely I didn't apply it correctly).  I did make progress with the "pip install -r requirements.txt" though. It threw a couple of errors, but I assume that is because I didn't have everything set up correctly.

kirberich, 2017-01-29:
Hey! So the instructions from the readme are for any linux (or mac) environment, but they do make some assumptions, like having virtualenv installed (on debian that's aptitude install python-virtualenv). After that you can run those instructions. You can also leave out the virtualenv, but then you'll need to run the `pip` command using `sudo`, otherwise you won't have permission to install the requirements.

It does seem however that there's something broken in the scipy package for debian, I ended up having to install it via the package manager instead of using pip.

However, as all of this is quite complicated, I've gone ahead and finally got the site back up. So if you'd rather not fiddle around with libraries for ages, go ahead and try http://solder-stencil.me/ again :)

REHolt, 2017-01-29:
Thanks, I thought I figured out how to make it work, but I was still seeing errors, so I'm glad you saw the same thing. I will try the online version, but I will likely keep fiddling with trying to get it working locally as well.

Thanks for the help.

Robert

REHolt, 2017-01-29:
Now that is REALLY COOL!!  already created STL's for both top and bottom and one is on the printer now, 11 min to print. can't wait to see how it turns out.

kirberich, 2017-01-29:
Glad to hear it! Be prepared for some fiddling though, you have to get the first layer extremely well calibrated for the stencil to come out well - a bit too far away and the stencil ends up not being solid, a bit too close and holes in the stencil don't print properly. But once you get it calibrated right I found the printed stencils to work almost as well as professional ones (as long as the parts aren't too small).

REHolt, 2017-01-29:
Yea, first print was a bust. I only got the stencil ledge for some reason. Just ran the files through Mesh Mixer and will try again. this time printing both at the same time.

kirberich, 2017-01-29:
How come you have stencils for top and bottom - are you planning on reflowing both sides? I've had cases where something was wrong with the STL file breaking the mesh, sometimes this happens when lines are exactly in the same place (if two holes overlap). Try to set the "increase hole size" option to something very small like 0.001 or so, that often fixes it.

Also if you like feel free to send me your gerber files, I'll try it out here and see if I can figure out what's wrong

REHolt, 2017-01-29:
Yes, the plan is to re-flow both sides. I was thinking the STL had mesh problems as well, so I used meshmixer to try and fix them. It didn't help. I'm thinking that my printer is set up with minimum number of layers, and two layers just won't cut it. I went in and created a custom profile and will see how that works. I'll also try your suggestion about the "increase hole size" option, if it doesn't work this time.

REHolt, 2017-01-29:
Same result, at this point I don't think it is your software. It looks good in meshmixer and on the build plate in the printer software, but it isn't printing the stencil layers, only the ledge. I'm going to check with the support guy for my printer and see if he can offer any suggestions.

kirberich, 2017-01-29:
If you want, send me over the stl or the gerber files I'll have a look if the files themselves look correct, just to rule out a problem in the software - you can upload them here or send them to robert@kirberich.de

REHolt, 2017-01-29:
Here is one of the STL files, I had to change the extension to .txt so that it would upload

[gts-4980848193.txt](https://github.com/kirberich/gerber_to_scad/files/737716/gts-4980848193.txt)

REHolt, 2017-01-29:
the problem does seem to be the printer. I changed the thickness to .3mm and now it is printing the stencil.   It may take some more fiddling though, as the first print is not very good.

kirberich, 2017-01-29:
Yep it looks fine for me - what works best for me btw is to print a single 0.15mm layer for the stencil - 0.1 (at least on my printer) is difficult to get solid, while two layers tend to bee too thick for the solder paste to release properly (most professional stencils seem to be about 0.1mm to 0.12mm thick).

[image]

REHolt, 2017-01-29:
Thanks, I'll give that a try. I appreciate the help getting this going. I am re-running the print now with a slightly different calibration. hopefully that should get the quality up.

kirberich, 2017-01-29:
Good luck, and no problem! 

For reference, here's what my printer spits out with a 0.15mm layer: (the top of the layer isn't great, it'd probably be good enough to use though)

[image]

REHolt, 2017-01-29:
Great resolution on the holes. Mine keeps filling some of them in. I may still have it too close. 

Here is my latest attempt. I didn't let it finish the ledge
[image]

REHolt, 2017-01-29:
I'm having a problem on the bottom stencil. The printer stops in one place and deposits a large amount of material, and then continues to print. It sort of looks like it is repeating a hole outline over and over. could this be due to holes being very close together? I tried increasing the hole size. Is there something else I can do to change this?

REHolt, 2017-01-29:
I re-ran everything from scratch starting with the cam file. Whatever the problem was it went away. it appears I'm getting a clean print. Must have just been a glitch.

REHolt, 2017-01-29:
One minor problem; the stencil for the bottom side is mirrored. the ledge is on the wrong side. Other than that, this last print was almost perfect,
[image]

kirberich, 2017-01-29:
That looks good! The few gaps in the stencil are probably fine, I figure this should be good to go. 

Very good point about the ledge, I've never tried printing a stencil for the bottom side, so this never occurred to me! You should be able to use it without a ledge (it's more of a bonus anyway), but I might try and spend a bit of time to allow reversing the direction of the ledge.

Hope the soldering will go well!

REHolt, 2017-01-29:
I modified my settings to hopefully take care of the gaps (at least some of them). I agree about the ledge, just getting this far is a huge bonus. I used both sides of the board to keep it small. My experience with reflow is that you can do each side separately without a problem. Once the parts are soldered on the first side, just turn it over and do the other. The biggest problem is  having something for it to set on to raise it up off of the surface. With the 3D printer I can make a custom holder for the board out of nylon, or even HIPS that will work great.

IF this works, it will save me some money since I wont have to buy a stencil.

REHolt, 2017-01-30:
It looks like the problem is solved.  I went back to the CAM file and there is an option to "Mirror" the output. I set the bottom for mirror and the stl file looks correct now. I will print it shortly to verify.

REHolt, 2017-02-06:
Built my custom board this weekend. The solder Stencil I printed, with the help of your program, was very helpful. The 0603 parts were much easier to place and re-flow solder rather than trying to solder by hand.

The Board actually, but I will be incorporating a couple of changes, including changing as much as possible to surface-mount parts. And you can be I will be printing another Stencil for this board as well as the second board I am going to build.

Thanks!

kirberich, 2017-02-06:
Awesome, I'm happy to hear that! I might add the mirroring option anyway, but it's cool that there's a workaround as well! (If I don't add it I might just add it to the hints section).

closed 2017-02-06 by kirberich