kirberich/gerber_to_scad #5: Got IndexError: tuple index out of range when run the script

HuyKhac, 2017-06-28:
Hi all,

I try to run the script with my *.gm1 and *.gtp gerber files to generate *.stl file for 3D printing.
However, I could not get it runs successfully.
Here is the log:
[Admins-MacBook-Pro:gerber_to_scad-master macOS$ python gerber_to_scad.py lora_dev_usb_poured.GM1 lora_dev_usb_poured.GTP output.stl
Traceback (most recent call last):
  File "gerber_to_scad.py", line 392, in <module>
    args.increase_hole_size
  File "gerber_to_scad.py", line 308, in process
    outline_shape = create_outline_shape(outline_file)
  File "gerber_to_scad.py", line 183, in create_outline_shape
    return convex_hull(outline_vertices)
  File "gerber_to_scad.py", line 23, in convex_hull
    hull = ConvexHull(points)
  File "scipy/spatial/qhull.pyx", line 2379, in scipy.spatial.qhull.ConvexHull.__init__ (scipy/spatial/qhull.c:24005)
IndexError: tuple index out of range

Can anyone tell me how to fix the problem?

Thanks,
HuyK

[gerbers.zip](https://github.com/kirberich/gerber_to_scad/files/1109099/gerbers.zip)

kirberich, 2017-06-28:
Hey! 

Thanks for reporting this :) I'm travelling right now, but I'll try and have a look at this on the weekend.

Can you tell me what software you used to create the schematic?

Cheers
Rob

HuyKhacNguyen, 2017-06-29:
I use Altium to create gerber files. Thanks

kirberich, 2022-06-18:
You probably don't much care about this board anymore 5 years later, but just for the sake of anyone else having a similar problem - if I try processing this board now it kind of works, but it breaks because the outline file doesn't actually contain an outline.

closed 2022-06-18 by kirberich