kirberich/gerber_to_scad #6: SCAD file preview is fine -> rendering contains only the ledge

benjaminaigner, 2018-08-23:
Dear Rob,

thx for this nice project, I would like to test a stencil from the 3D printer :-).

Anyway, there seems to be a problem with the scad file:

* I tried the online page -> error (cannot create STL file)
* Python script works like a charm, no problems with pip or the script itself
* Opening the SCAD file in OpenSCAD (v2015.03-2  on Debian testing) works
too
* A preview is rendered correctly, I can see all parts/details of the stencil
* If a render it (necessary to export as STL file), there is only the ledge left

Do you have any advice how to proceed?

I've attached 2 different board files I tried.
Both of them are exported from KiCAD5.

PS: No gbr files in GitHUB -> renamed them to .txt
[ble_power_plug_relais_kicad-F.Paste.txt](https://github.com/kirberich/gerber_to_scad/files/2313884/ble_power_plug_relais_kicad-F.Paste.txt)
[ble_power_plug_relais_kicad-Edge.Cuts.txt](https://github.com/kirberich/gerber_to_scad/files/2313885/ble_power_plug_relais_kicad-Edge.Cuts.txt)
[MPPT_ARM-Edge.Cuts.txt](https://github.com/kirberich/gerber_to_scad/files/2313889/MPPT_ARM-Edge.Cuts.txt)
[MPPT_ARM-F.Paste.txt](https://github.com/kirberich/gerber_to_scad/files/2313890/MPPT_ARM-F.Paste.txt)

kirberich, 2018-08-23:
Hi! I've had a quick look and the files seem alright, so it's probably a bug having to do with unsupported gerber commands. I'll have a look tonight to see if I can fix it :)

benjaminaigner, 2018-08-28:
@kirberich 

Thank you very much :-)

kirberich, 2022-06-18:
I couldn't solve this problem back then, but if you're still curious, the boards work fine now, 4 years later :D 
[image]

closed 2022-06-18 by kirberich