kirberich/gerber_to_scad #11: TypeError: offset_points() got an unexpected keyword argument 'inside'

W-M-D, 2021-04-20:
got this error while running 

/home/matt/Projects/Code/gerber_to_scad/./gerber_to_scad.py:442: DeprecationWarning: 'U' mode is deprecated
  outline_file = open(args.outline_file, 'rU')
/home/matt/Projects/Code/gerber_to_scad/./gerber_to_scad.py:443: DeprecationWarning: 'U' mode is deprecated
  solderpaste_file = open(args.solderpaste_file, 'rU')
Traceback (most recent call last):
  File "/home/matt/Projects/Code/gerber_to_scad/./gerber_to_scad.py", line 449, in <module>
    process(
  File "/home/matt/Projects/Code/gerber_to_scad/./gerber_to_scad.py", line 383, in process
    ledge_shape = offset_shape(outline_shape, 1.2)
  File "/home/matt/Projects/Code/gerber_to_scad/./gerber_to_scad.py", line 211, in offset_shape
    offset_3d_points = utils.offset_points(
TypeError: offset_points() got an unexpected keyword argument 'inside'

dimonDDL, 2021-05-16:
Got the exact same issue as well.

TypeError: offset_points() got an unexpected keyword argument 'inside'

kirberich, 2022-06-18:
Hi, sorry for this very late reply, and thank you for reporting! I think this got fixed a while ago, but let me see if you're still seeing this problem

closed 2022-06-18 by kirberich