kirberich/gerber_to_scad #16: ModuleNotFoundError: No module named 'gerber'

braxtron, 2023-01-05:
Running Poetry via Python 3.10.9 from PowerShell on Win10, every time I run python main.py (regardless of any flags or input files), I get:

python .\main.py
Traceback (most recent call last):
  File "C:\gerber_to_scad\main.py", line 2, in <module> from gerber_to_scad import process_gerber
  File "C:\gerber_to_scad\gerber_to_scad\__init__.py", line 1, in <module> from .conversion import process_gerber
  File "C:\gerber_to_scad\gerber_to_scad\conversion.py", line 5, in <module> from gerber import primitives
ModuleNotFoundError: No module named 'gerber'

Gadgeteering, 2023-04-23:
You are missing some of the libraries

I ended up using the `pip install -r requirements.txt` command using the requirements.txt file below as I couldn't get `poetry install` to work

```
Django == 3.2.7
django-bootstrap3 ==15.0.0
django-environ ==0.7.0
gunicorn ==20.1.0
pcb-tools ==0.1.6
scipy ==1.7.1
solidpython ==1.1.1
solid 
taskipy ==1.9.0
black ==22.3.0

kirberich, 2023-04-23:
Hi @braxtron sorry for only responding now, for some reason I didn't see this issue! 

As @Gadgeteering points out you have to install the libraries first, you can find the general instructions in the readme!

@Gadgeteering can you let me know what issues you were having with poetry - was it just that you couldn't get poetry installed, or even with poetry installed the `poetry install` command wouldn't work?

closed 2024-02-10 by kirberich