kirberich/gerber_to_scad #20: TypeError: 'NoneType' object is not subscriptable conversion.py  line 290

Gadgeteering, 2023-04-23:
TypeError: 'NoneType' object is not subscriptable -> gerber_to_scad/gerber_to_scad/conversion.py @ line 290, in <listcomp>  V(p[0], p[1])

There seems to be a problem with the Solid Python offset  function

offset_points function returns none SolidCode/SolidPython#206

kirberich, 2023-04-23:
Can you send me the files you're trying to process so I can try and replicate this?

Gadgeteering, 2023-04-23:
Files attached

[Archive.zip](https://github.com/kirberich/gerber_to_scad/files/11303674/Archive.zip)

kirberich, 2023-04-23:
Hm this works fine for me, both locally and on https://solder-stencil.me

Here's the output I get:
[image]

As you said you couldn't get poetry to work and manually generated a list of depenencies, I think something might not be quite right in your local setup - I'd recommend trying to get poetry to work properly

If you can't make that work, here's a requirements.txt for windows generated from the actual current dependencies:

```
asgiref==3.5.2 
cairocffi==0.9.0 
cffi==1.15.0 
colorama==0.4.5 
django-bootstrap3==15.0.0 
django-environ==0.7.0 
django==3.2.13 
euclid3==0.01 
gunicorn==20.1.0 
mslex==0.3.0
numpy==1.22.4 
pcb-tools==0.1.6 
ply==3.11 
prettytable==0.7.2 
psutil==5.9.1 
pycparser==2.21 
pypng==0.0.19 
pytz==2022.1 
scipy==1.8.1 
setuptools==65.6.0 
solidpython==1.1.3 
sqlparse==0.4.2 
taskipy==1.10.2 
tomli==2.0.1 
```

kirberich, 2023-04-23:
Ah sorry, ignore the above! I didn't see the extra options you added, with that I get the error as well. Lemme have a look into it!

kirberich, 2023-04-23:
Alrighty found the problem - there was an issue with how points were compared, which in another part of the code that combines shapes together caused some duplicated points within shapes. Solidpython then got confused by the zero-length line segments between the duplicated points and returned those None values.

There's still a related bug/inconsistency in solidpython - either `offset_points` should raise an error when it cannot offset the points (probably the right answer) or at least the type hint should indicate that it can return None values - but the main problem here was just how vectors got compared in my code.

The website is updated, so this should all be working fine for now - though I wouldn't recommend using those actual settings you pasted, as the holes will all merge together.

closed 2023-04-23 by kirberich

Gadgeteering, 2023-04-23:
Thanks for the quick response