kirberich/gerber_to_scad #21: does this support gerber from KiCAD? I don't see any .gko file

Utopiah, 2024-02-07:
Hi, neat project, wondering if I can use it with KiCAD though. I exported my project and get : 

kicad-like-a-pro-3rd-LED-torch-B_Cu.gbl
kicad-like-a-pro-3rd-LED-torch-B_Mask.gbs
kicad-like-a-pro-3rd-LED-torch-B_Paste.gbp
kicad-like-a-pro-3rd-LED-torch-B_Silkscreen.gbo
kicad-like-a-pro-3rd-LED-torch-Edge_Cuts.gm1
kicad-like-a-pro-3rd-LED-torch-F_Cu.gtl
kicad-like-a-pro-3rd-LED-torch-F_Mask.gts
kicad-like-a-pro-3rd-LED-torch-F_Paste.gtp
kicad-like-a-pro-3rd-LED-torch-F_Silkscreen.gto
kicad-like-a-pro-3rd-LED-torch-job.gbrjob
kicad-like-a-pro-3rd-LED-torch-NPTH.drl
kicad-like-a-pro-3rd-LED-torch-PTH.drl

so wondering if something could replace the .gko file in the example.

kirberich, 2024-02-10:
Hi! Yeah the app does support kicad (as much as it supports anything, mileage varies greatly between projects)

Kicad seems to use different file extensions sometimes or maybe the versions have changed, but the important thing is to use the outline file and the paste file - in your example I believe it'd be `kicad-like-a-pro-3rd-LED-torch-Edge_Cuts.gm1` and `F_Paste` / `B_Paste`

closed 2024-10-14 by kirberich

Utopiah, 2024-10-14:
Thank you for the work and the clarification!

kirberich, 2024-10-15:
No worries, let me know if using kicad files gives you any trouble!